RobustFieldAutonomyLab/LeGO-LOAM #160: roslaunch error

abish-a, 2020-01-28:
w.r.t issue #142. I also get the similar error (see below). I installed the gtsam4.0.0-alpha2 as mentioned. 
Do you know how can I fix that?
Thanks

```
process[base_link_to_camera-4]: started with pid [959]
process[imageProjection-5]: started with pid [961]
process[featureAssociation-6]: started with pid [970]
process[mapOptmization-7]: started with pid [972]
/home/abi/catkin_ws/devel/lib/lego_loam/mapOptmization: error while loading shared libraries: libmetis.so: cannot open shared object file: No such file or directory
process[transformFusion-8]: started with pid [974]
[ INFO] [1580219166.650001857]: ----> Transform Fusion Started.
[mapOptmization-7] process has died [pid 972, exit code 127, cmd /home/abi/catkin_ws/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/abi/.ros/log/87d890c6-41d4-11ea-9780-ac1f6b46a81c/mapOptmization-7.log].
log file: /home/abi/.ros/log/87d890c6-41d4-11ea-9780-ac1f6b46a81c/mapOptmization-7*.log
```

TixiaoShan, 2020-01-28:
It's very likely related to GTSAM. You can put some ```cout``` in the code to see when the program dies. I suspect the code will die [here](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/38f087c06b8832c49760d0de54ce74321f5bcc3f/LeGO-LOAM/src/mapOptmization.cpp#L345).

abish-a, 2020-01-28:
libmetis library was not installed. Fixed it by installing **libparmetis-dev**

closed 2020-01-28 by abish-a

LJZcn, 2021-09-15:
sudo apt-get install libparmetis-dev

aladin2022, 2022-02-24:
hi,I'am trying to run the lego loam package using gtsam 4.1.1 , ubunto 20.04.03 , i had successfully built the project and installed the libmetis library but still have the same error. 
[mapOptmization-7] process has died [pid 28534, exit code 127, cmd /home/aladin/outdoorslam_ws/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/aladin/.ros/log/9e0ac2b2-9559-11ec-a76b-05c4747b78f6/mapOptmization-7.log].
log file: /home/aladin/.ros/log/9e0ac2b2-9559-11ec-a76b-05c4747b78f6/mapOptmization-7*.log

TheSeanParker, 2022-06-05:
after libmetis library was not installed. Fixed it by installing libparmetis-dev
my code has the  problem:
/home/code/Videos/LEGO_LOAM_Description/devel/lib/lego_loam/mapOptmization: symbol lookup error: /home/code/Videos/LEGO_LOAM_Description/devel/lib/lego_loam/mapOptmization: undefined symbol: _ZNK5gtsam16NoiseModelFactor5printERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKN5boost8functionIFS6_mEEE
[mapOptmization-6] process has died [pid 9647, exit code 127, cmd /home/code/Videos/LEGO_LOAM_Description/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/code/.ros/log/861c9594-e46a-11ec-b80b-d4d252ed3c76/mapOptmization-6.log].
log file: /home/code/.ros/log/861c9594-e46a-11ec-b80b-d4d252ed3c76/mapOptmization-6*.log
[ WARN] [1654394151.255441849, 1497707651.261446535]: Detected jump back in time of 59.9663s. Clearing TF buffer.

Waydelucien, 2023-10-10:
> after libmetis library was not installed. Fixed it by installing libparmetis-dev my code has the problem: /home/code/Videos/LEGO_LOAM_Description/devel/lib/lego_loam/mapOptmization: symbol lookup [...]

I have the same problem. It shows as the following:
[mapOptmization-6] process has died [pid 6005, exit code 127, cmd /home/yang/path_planner_paper/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/yang/.ros/log/222299f0-62a2-11ee-b0a9-18c04dce611c/mapOptmization-6.log].
log file: /home/yang/.ros/log/222299f0-62a2-11ee-b0a9-18c04dce611c/mapOptmization-6*.log
But I see the program can still run. Do you solve it?