RobustFieldAutonomyLab/LeGO-LOAM #162: Parallelising LeGO-LOAM

lsdta, 2020-02-14:
Hi @TixiaoShan ,

Have you attempted to parallelise your mapping algorithm at all? I see you have used the NVIDIA Jetson TX2, but was the GPU used at all? Is there any reason why a GPU board is useful, or have you found that LeGO-LOAM is inherently mostly sequential? (and thus CPU-dominant). [It seems](https://github.com/nischalkn/gpuSLAM) that SLAM can be improved with a GPU, but by the looks of your code it is just running on the CPU?

Any ideas appreciated, I am planning on running this on an embedded board but not sure if it is worth utilising a GPU-accelerated board or not.

TixiaoShan, 2020-02-15:
@lsdta 
Yes, all the code runs on the CPU, GPU is not used at all. The part that requires the most computational resources in the code is the optimization, which is [this function](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/38f087c06b8832c49760d0de54ce74321f5bcc3f/LeGO-LOAM/src/mapOptmization.cpp#L1303). But I am not sure how easy it is to parallelize it.

lsdta, 2020-02-15:
So buying a GPU-accelerated board is not worth it if all it's running is LeGO-LOAM without visualisation? 

Have you timed your code to find out that `scan2MapOptimization` is the computationally intensive part? Do you know if it is the `corner`, `surf` or `LM` optimisation sections that is the worst? 

Utilising the GPU with CUDA (e.g. on the Jetson TX2) may be worth the effort, the matrix operations would be possible to parallelise. However, I am not sure how many dependencies there are in your for loops that may limit the extent of parallelisation.

TixiaoShan, 2020-02-16:
That's right, LeGO-LOAM doesn't need GPU without visualization. I did time the code, the results are in the code. I will take a closer look at those sections when I have time.

lsdta, 2020-02-19:
Thank you. I will focus on a powerful CPU rather than GPU for SLAM then.

closed 2020-02-19 by lsdta