RobustFieldAutonomyLab/LeGO-LOAM #163: How to integrate IMU and Point cloud Data of Ouster in the Algorithm?

adityarath302, 2020-02-17:
I have been trying to use LeGo-LOAM for Ouster OS1-32 and was successful to get a 3D Point Cloud.
I want to know how to integrate the IMU with the algorithm.

TixiaoShan, 2020-02-17:
Hi @adityarath302 
Currently, LeGO-LOAM is not compatible with a 6-axis IMU because it needs roll, pitch, and yaw information to correct the pose. I think [lio-mapping](https://github.com/hyye/lio-mapping) works with Ouster lidar as it works with a 6-axis IMU.

adityarath302, 2020-02-17:
Thank you so much @TixiaoShan

closed 2020-02-18 by TixiaoShan

L-Reichardt, 2022-07-27:
@TixiaoShan Great work on LegoLoam, I've been using it successfully with an OS1-64. Does this incompatibility with the 6-axis IMU still exist or was this ever changed?

falak123-sudo, 2023-08-26:
I have been trying to use lego -LOAM for ouster os1 32 lidar to get 3d point cloud can you please me an idea how to acheive it.

MohammadAmmargk8497, 2023-11-24:
I have been getting only a single point cloud scan. How do I fix it