RobustFieldAutonomyLab/LeGO-LOAM #165: Cmake not working

adityarath302, 2020-03-02:
Scanning dependencies of target topic_tools_generate_messages_nodejs
[ 46%] Built target topic_tools_generate_messages_nodejs
Scanning dependencies of target actionlib_generate_messages_py
[ 46%] Built target actionlib_generate_messages_py
Scanning dependencies of target pcl_ros_gencfg
[ 46%] Built target pcl_ros_gencfg
Scanning dependencies of target topic_tools_generate_messages_cpp
[ 46%] Built target topic_tools_generate_messages_cpp
Scanning dependencies of target rosgraph_msgs_generate_messages_cpp
[ 46%] Built target rosgraph_msgs_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_generate_messages_cpp
[ 46%] Built target dynamic_reconfigure_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_generate_messages_eus
[ 46%] Built target dynamic_reconfigure_generate_messages_eus
Scanning dependencies of target pcl_msgs_generate_messages_cpp
[ 46%] Built target pcl_msgs_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_generate_messages_lisp
[ 46%] Built target dynamic_reconfigure_generate_messages_lisp
Scanning dependencies of target featureAssociation
[ 53%] Building CXX object LeGO-LOAM/LeGO-LOAM/CMakeFiles/featureAssociation.dir/src/featureAssociation.cpp.o
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52:0,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/enercomp/ll_ws/src/LeGO-LOAM/LeGO-LOAM/include/utility.h:24,
                 from /home/enercomp/ll_ws/src/LeGO-LOAM/LeGO-LOAM/src/featureAssociation.cpp:35:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libGL.so', needed by '/home/enercomp/ll_ws/devel/lib/lego_loam/featureAssociation'.  Stop.
CMakeFiles/Makefile2:1718: recipe for target 'LeGO-LOAM/LeGO-LOAM/CMakeFiles/featureAssociation.dir/all' failed
make[1]: *** [LeGO-LOAM/LeGO-LOAM/CMakeFiles/featureAssociation.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

adityarath302, 2020-03-02:
How to fix this issue?

TixiaoShan, 2020-03-02:
Hi @adityarath302 
We never met this issue before. Have you tried to follow the tutorial to install [here](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM)?

adityarath302, 2020-03-03:
After some tweaking this is the response

[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  6%] Built target cloud_msgs_generate_messages_cpp
[ 13%] Built target cloud_msgs_generate_messages_nodejs
[ 26%] Built target cloud_msgs_generate_messages_py
[ 33%] Built target cloud_msgs_generate_messages_lisp
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libGL.so', needed by '/home/enercomp/test_ws/devel/lib/lego_loam/transformFusion'.  Stop.
CMakeFiles/Makefile2:3486: recipe for target 'LeGO-LOAM/LeGO-LOAM/CMakeFiles/transformFusion.dir/all' failed
make[1]: *** [LeGO-LOAM/LeGO-LOAM/CMakeFiles/transformFusion.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libGL.so', needed by '/home/enercomp/test_ws/devel/lib/lego_loam/mapOptmization'.  Stop.
CMakeFiles/Makefile2:2331: recipe for target 'LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/all' failed
make[1]: *** [LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/all] Error 2
[ 46%] Built target cloud_msgs_generate_messages_eus
[ 46%] Built target cloud_msgs_gencpp
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed

adityarath302, 2020-03-03:
I solved it! It was some linkage error due to upgrade of libGL.so file upgrade to libGL.so.1.0.0

TixiaoShan, 2020-03-04:
Glad you fixed it!

closed 2020-03-04 by TixiaoShan

suchet4620, 2020-05-11:
@adityarath302 I just faced the same problem, could you please guide me more details how to solve the problem? (I am not quite sure that I also upgraded libGL.so or not)

YHuniv, 2023-01-24:
here ho to resolve the problem :                                                                                                                                              sudo mv /usr/lib/x86_64-linux-gnu/libGL.so.1.0.0 /usr/lib/x86_64-linux-gnu/libGL.so