RobustFieldAutonomyLab/LeGO-LOAM #166: how to compute the roll pitch and yaw of the IMU

WEICHUAN1107, 2020-03-05:
First of all, thank you for your contribution. I'm still confused with the roll, ptich and yaw.
is it the angle between the current imu frame and the initial moment imu frame, or the angle between the current imu frame and the east-north-up frame?

Thank you and I'm looking forward to your reply.

TixiaoShan, 2020-03-05:
"angle between the current imu frame and the east-north-up frame" should be the right answer.

WEICHUAN1107, 2020-03-06:
Thanks for your @TixiaoShan reply!
I'm also confused if the x-axis of the radar and imu coordinate system is forward, the y-axis is left, and the z-axis is up?

TixiaoShan, 2020-03-06:
> Thanks for your @TixiaoShan reply!
> I'm also confused if the x-axis of the radar and imu coordinate system is forward, the y-axis is left, and the z-axis is up?

That is correct. One thing you have to keep in mind is that LOAM converts the point cloud from lidar frame to camera frame in [here](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/f61b97f200e4612022833c6be19d34cbcd055871/LeGO-LOAM/src/featureAssociation.cpp#L487). I think the reason behind this is Ji Zhang was also developing V-LOAM at that time. This conversion makes his LOAM code work seamlessly with V-LOAM.

WEICHUAN1107, 2020-03-07:
> That is correct. One thing you have to keep in mind is that LOAM converts the point cloud from lidar frame to camera frame in [...]

Thanks very much! I got it

closed 2020-03-07 by TixiaoShan