RobustFieldAutonomyLab/LeGO-LOAM #172: Some questions about loop closure detection

zcg3648806, 2020-04-06:
extern const float surroundingKeyframeSearchRadius = 50.0; // key frame that is within n meters from current pose will be considerd for scan-to-map optimization (when loop closure disabled)
extern const int   surroundingKeyframeSearchNum = 50; // submap size (when loop closure enabled)
// history key frames (history submap for loop closure)
extern const float historyKeyframeSearchRadius = 7.0; // key frame that is within n meters from current pose will be considerd for loop closure
extern const int historyKeyframeSearchNum = 25; // 2n+1 number of hostory key frames will be fused into a submap for loop closure

If I want to make the loop closure detection as good as possible, which of the above parameters should be increased? and which should be reduced? Thank you!

YushengWHU, 2020-04-11:
for my case, I just reduced the historyKeyframeSearchRadius from 7.0 to 5.0.
Sensor tested: Hesai 40 channel Lidar
Speed:             around 50km/h

kissb2, 2020-04-11:
BTW fyi, 
I integrated the global loop detector (scan context) into the LeGO-LOAM pipeline
https://github.com/irapkaist/SC-LeGO-LOAM 
This ad-hook global loop detector works even large drift exist (e.g., https://www.youtube.com/watch?v=p-NsVs8GATA&feature=youtu.be&t=436)
and could enhance LeGO-LOAM's original radius search-based loop detection.

zcg3648806, 2020-04-13:
@YushengWHU  @kissb2  Thank you for your reply！

closed 2020-05-20 by stale[bot]