RobustFieldAutonomyLab/LeGO-LOAM #173: Hi, why i cannot got .pcd map after I use Ctrl+C to stop the program? I did't see any .pcd file in /tmp/

DeepDuke, 2020-04-06:


YushengWHU, 2020-04-11:
That's because you didn't set the correct saving directory.
You can do it by changing the code in MapOptimization, column 704 if I remember correctly.
Also, you can write you own publish node, like a /save_pcd topic, then you can pub the cloud map whenever you wish.

DeepDuke, 2020-04-20:
I changed it, but still canot make it work. Ok, I saved map from another topic finally.

closed 2020-04-20 by DeepDuke

nyangshawbin, 2020-06-03:
@DeepDuke 
Hi, could you give me some advice on how to save the map from another topic? I made some changes to the utility.h as well as mapOptimization.cpp as mentioned in #88. 

I am running lego loam on a rosbag. Is the .pcd file supposed to be generated after i terminate Lego-loam?

DeepDuke, 2020-07-31:
@nyangshawbin hi, you can record the last seconds of topic `laser_cloud_surround`,  and then convert the .bag file  into .pcd by running `rosrun pcl_ros bag_to_pcd xxx.bag`