RobustFieldAutonomyLab/LeGO-LOAM #174: project the odometry information on the google's map in real time

JrDarwin, 2020-04-12:
Hi @TixiaoShan ,
      Can you please tell me how to project the odometry information on the google's map in real time?It's like upper right corner of your 'demo.gif'. I made some small changes to the algorithm and wanted to do a real-time verification. 
      Thank you very much for your kind consideration and I'm looking forward to your early reply.

TixiaoShan, 2020-04-13:
> Hi @TixiaoShan ,
> Can you please tell me how to project the odometry information on the google's map in real time?It's like upper right corner of your 'demo.gif'. I made some small changes to the [...]

@JrDarwin 
Hi, I used Final Cut Pro to show the trajectory over a google map image.

JrDarwin, 2020-04-13:
> @JrDarwin
> Hi, I used Final Cut Pro to show the trajectory over a google map image.

@TixiaoShan 
thanks I'll give that a try.

closed 2020-04-13 by JrDarwin