RobustFieldAutonomyLab/LeGO-LOAM #175: how to use kitti datasets （vlp HDL-64E）for test?

shikeqin123, 2020-04-13:
I have a problem to transform 64 line pointclouds to 16 line pointclouds.

bryantaoli, 2020-05-01:
> I have a problem to transform 64 lines pointclouds to 16 lines pointclouds.

I also have this problem, can you deal with it？

shikeqin123, 2020-05-01:
> > I have a problem to transform 64 lines pointclouds to 16 lines pointclouds.
> 
> I also have this problem, can you deal with it？

解决不了，kitti保存点云的方式是无序的，并不是一线一线存储，你可以算算单个点云的水平角和垂直角来看看

bryantaoli, 2020-05-01:
> > > I have a problem to transform 64 lines pointclouds to 16 lines pointclouds.
> > 
> > 
> > I also have this problem, can you deal with it？
> 
> 解决不了，kitti保存点云的方式是无序的，并不是一线一线存储，你可以算算单个点云的水平角和垂直角来看看

这说明kitti没办法跑lego_loam吗？

shikeqin123, 2020-05-01:
> > > I have a problem to transform 64 lines pointclouds to 16 lines pointclouds.
> > 
> > 
> > I also have this problem, can you deal with it？
> 
> 解决不了，kitti保存点云的方式是无序的，并不是一线一线存储，你可以算算单个点云的水平角和垂直角来看看

> > > > I have a problem to transform 64 lines pointclouds to 16 lines pointclouds.
> > > 
> > > 
> > > I also have this problem, can you deal with it？
> > 
> > 
> > 解决不了，kitti保存点云的方式是无序的，并不是一线一线存储，你可以算算单个点云的水平角和垂直角来看看
> 
> 这说明kitti没办法跑lego_loam吗？

论文里用了kitti,应该需要特殊处理，作者也没说，就在readme提了一句

bryantaoli, 2020-05-01:
> 论文里用了kitti,应该需要特殊处理，作者也没说，就在readme提了一句

ok，那这个开源也开的不是很准确啊

shikeqin123, 2020-05-01:
> ok，那这个开源也开的不是很准确啊

https://zhuanlan.zhihu.com/p/71731356 这个你看看

shikeqin123, 2020-05-01:
> ok，那这个开源也开的不是很准确啊

作者提供了他自己采的vlp16的数据集，需要翻墙下载

bryantaoli, 2020-05-01:
> 作者提供了他自己采的vlp16的数据集，需要翻墙下载
老哥有链接吗

shikeqin123, 2020-05-01:
> > 作者提供了他自己采的vlp16的数据集，需要翻墙下载
> > 老哥有链接吗

https://github.com/TixiaoShan/Stevens-VLP16-Dataset

bryantaoli, 2020-05-01:
> 作者提供了他自己采的vlp16的数据集，需要翻墙下载

https://github.com/TixiaoShan/Stevens-VLP16-Dataset ，这个数据没啥用啊，只有激光，我想融合别的传感器还是得找别的数据

shikeqin123, 2020-05-01:
> https://github.com/TixiaoShan/Stevens-VLP16-Dataset ，这个数据没啥用啊，只有激光，我想融合别的传感器还是得找别的数据

那只能找找其他数据集了。。或者自己采

bryantaoli, 2020-05-01:
> 那只能找找其他数据集了。。或者自己采

kitti激光这个也太差了吧，太惨了，还得自己修理

shikeqin123, 2020-05-01:
我已放弃使用kitti来整这个算法

bryantaoli, 2020-05-01:
> 我已放弃使用kitti来整这个算法
> […](#)

https://github.com/Mitchell-Lee-93/kitti-lego-loam

bryantaoli, 2020-05-01:
https://github.com/Mitchell-Lee-93/kitti-lego-loam

shikeqin123, 2020-05-01:
> https://github.com/Mitchell-Lee-93/kitti-lego-loam
good job

closed 2020-06-07 by stale[bot]

ZhangZh3ng, 2021-12-01:
> 
请问你都在哪些数据集上验证了该算法