RobustFieldAutonomyLab/LeGO-LOAM #176: 2d map with Lego_loam

mohamadmahdavian, 2020-04-15:
Hi 

Is there anyway that I could get 2d map with lego_loam package? I currently can get 3d, but how can I get 2d map with it?

Thank You

TixiaoShan, 2020-04-16:
One way of doing it is to convert the 3D point cloud into 2D laser scan, then feed it to gmapping. But it's kind of convoluted.

closed 2020-05-23 by stale[bot]

getupgetup, 2020-06-02:
> Hi
> 
> Is there anyway that I could get 2d map with lego_loam package? I currently can get 3d, but how can I get 2d map with it?
> 
> Thank You

Maybe you can refer class XRayPointsProcessor in google's cartographer, which provide a solution to get a 2d xray map of 3d pointcloud.

HarshalDawar, 2020-07-01:
I found [octomap_server](http://wiki.ros.org/octomap_server) to be useful in this regard.