RobustFieldAutonomyLab/LeGO-LOAM #177: Doesnot work on own rosbag data

vijaygovindappa, 2020-04-16:
Hi All, 
First of all, you all have doen amazing job. From the video the results looks really good. 
Apologies, if my questions are silly, i am new to SLAM. 
I am trying to run on my own rosbag data and results doesnot look great. I believe it is because of the IMU. 
I have 2 questions. 
1. When you say IMU and LiDAR should be aligned, does that mean the axis should be aligned, or the pose should be aligned as in the x, y,z of IMU and LiDAR shoud be same ? 
2. Should the frequency of IMU and LiDAR match ? 

Waiting for your reply. 

Thank you

vijaygovindappa, 2020-04-16:
My LiDAR is VLP32-E running at 16hz and IMu is running at 50 hz.

bugerry87, 2020-04-20:
Try this configuration in utility.h

```
// VLP-32c
extern const int N_SCAN = 32;
extern const int Horizon_SCAN = 1800;
extern const float ang_res_x = 0.2;
extern const float ang_res_y = 40.0/float(N_SCAN-1);
extern const float ang_bottom = 25.0;
extern const int groundScanInd = 20;
```

**Note:** I assume you meant VLP-32c

closed 2020-05-27 by stale[bot]