RobustFieldAutonomyLab/LeGO-LOAM #181: Map drifts over time

saharLeisiazar, 2020-05-13:
Hi
I am using lego loam with ouster lidar and xsens imu and my problem is: when I run lego loam, although my robot is steady, map and trajectory drift over time. Is there any parameter I can change or tune?

s271, 2020-05-21:
Have you tried pure ouster lidar without imu? May be xsens imu is not synchronized with lidar good enough.

saharLeisiazar, 2020-05-21:
> Have you tried pure ouster lidar without imu? May be xsens imu is not properly synchronized or not properly aligned with lidar. I got good result on ouster 64 without imu.

Thanks for responding. I tried it with just lidar, still drifts. I checked it with many data recorded in different places. I noticed that at large and open places, lego loam works well but at small places like office it drifts. Is there any parameter I can tune to work for all places?

shikeqin123, 2020-05-22:
you are right,I try lego-loam in room failed also.I think problem is there is not enough features in room.

cafgiu, 2020-05-28:
Hi all. I am using lego loam with VLP16 lidar without IMU in indoor environment and I have the same problem as @saharLeisiazar. I builded the lego loam on the NVIDIA Jetson Nano Board. 

Do you have any suggestions?

Thank you.

zorosmith, 2020-06-09:
maybe u can use ALOAM in indoor environment.
it works well.

On 05/28/2020 17:55, cafgiu wrote:

Hi all. I am using lego loam with VLP16 lidar without IMU in indoor environment and I have the same problem as @saharLeisiazar. I builded the lego loam on the NVIDIA Jetson Nano Board.

Do you have any suggestions?

Thank you.

closed 2020-07-17 by stale[bot]

YiChenCityU, 2020-07-28:
> Hi
> I am using lego loam with ouster lidar and xsens imu and my problem is: when I run lego loam, although my robot is steady, map and trajectory drift over time along x and y. Is there any parameter I can change or tune?

Hi, I am also using ouster-64 to run lego-loam. But the rviz was not display right. Do you know how to fix it?
[image]

Garrus007, 2021-01-03:
@YiChenCityU, I guess I have same issue. Check RViz log in the terminal, do you have something like this?
```
Invalid argument "/camera" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like
```

I solved this issue by removing all leading "/" in the tf frame names. Just run the following command for quick replace:
```bash
cd <LeGO-LOAM source dir>
find -name "*.cpp" | xargs -I{} sed 's/frame_id = "\//frame_id = "/g' -i {}
```