RobustFieldAutonomyLab/LeGO-LOAM #183: saved map is in z x plane

gnykumar, 2020-05-22:
Hi,

I am trying out LeGO-LOAM with different dataset,
the thing i observed is map is generating in z x plane
even i seen same in only LOAM, means
x value is in z 
y value is in x
z value is in y
and i check in integrated_to_init topic which publish odometry data
same result.

I want map to be in xy plane, for this where i should change,
or is there any other package with this requirement.

It will be great If i get a solution for this in soon!

[image]

gaohongfein, 2020-06-01:
have you solved it?

getupgetup, 2020-06-02:
I think there's an easy way if you don't want to change the source code. You can generate another rosbag which all points's coordinate are rotated by yourself. For example, point(1, 2, 3)-->point(3, 1, 2)

Yeah2333, 2020-07-01:
In this,you can add transform code in mapOptimaziton node,such as 
` Eigen::Matrix4f transform_1 = Eigen::Matrix4f::Identity();
            float theta = PI/2.0;

            transform_1 (1,1) = cos (theta);
            transform_1 (1,2) = -sin(theta);
            transform_1 (2,1) = sin (theta);
            transform_1 (2,2) = cos (theta);
            Eigen::Matrix4f transform_2 = Eigen::Matrix4f::Identity();
            transform_2 (0,0) = cos (theta);
            transform_2 (0,1) = -sin(theta);
            transform_2 (1,0) = sin (theta);
            transform_2 (1,1) = cos (theta);
            Eigen::Matrix4f transform_3 = transform_2*transform_1;

            pcl::PointCloud<PointType>::Ptr transformed_cloud(new pcl::PointCloud<PointType>());
            pcl::transformPointCloud(*globalMapKeyFrames,*transformed_cloud,transform_3);
            pcl::io::savePCDFile(pcd_path,*transformed_cloud);`

closed 2020-08-08 by stale[bot]

aprilliuwei, 2021-12-09:
hello, do you solve this question?

gnykumar, 2021-12-13:
Sorry for the late reply to all. 
Yes, By multiplying with proper rpy values to correct the data. Basically it is rotated.  If I am not wrong I gave pitch of -90 degree to bring it back. I have done this long back. So don't remember exactly about it roll, pitch or yaw I gave. But it rotation I just know.
We can do either before process of each scan or after also.