RobustFieldAutonomyLab/LeGO-LOAM #185: How to change the utility.h when using lidar Ouster 64?

zwmisdu, 2020-06-05:
In the newset code, the parameter 'useCloudRing'  is used for velodyne which has the channel 'ring' .From the readme I know that Ouster has the same channel named 'r'. After modifing the parameter 'ring' in utility.h and imageProjection.cpp, the program can be complied successfully but when run the program, there is the warning that 'Failed to find match for field 'r'. So what should I do to run LEGO-LOAM with Ouster64 ? (I have used the config ‘ Ouster OS1-64’)

TixiaoShan, 2020-06-06:
@zwmisdu 
Have you tried to echo your point cloud message and see if there is field "r" in it?

zwmisdu, 2020-06-08:
> @zwmisdu
> Have you tried to echo your point cloud message and see if there is field "r" in it?

Thanks for your reminding and I check the point cloud message. I find that there is a filed named 'ring' in the data from OUSTER 64, so I change the code back to 'ring', but still mentioned 'Failed to find match for field 'ring'. Finally, I choose to set the parameter 'useCloudRing' to false which can run the algorithm succesfully. 
Maybe ……Could you tell the advantage when we set the parameter 'useCloudRing' to be true? I think this parameter can just replace the work to calculate the rowIdn.
'

TixiaoShan, 2020-06-08:
@zwmisdu 
Set it to true can accommodate sensors that have non-linear distributed channels, i.e. VLP-32C. It also slightly reduces computations.

zwmisdu, 2020-06-08:
> @zwmisdu
> Set it to true can accommodate sensors that have non-linear distributed channels, i.e. VLP-32C. It also slightly reduces computations.

Okay I see.Thanks for your reply and I will try to change my dataset and do more work.

closed 2020-06-08 by zwmisdu

7zhoulei, 2020-06-25:
> 'Failed to find match for field 'ring
Hello, I am testing Lego-loam using KITTI dataset. KITTI dataset use HDL-64E. I convert KITTI dataset to .bag file.  Could you please tell me how to check the point cloud message to see if there is a "ring" field or similar field?