RobustFieldAutonomyLab/LeGO-LOAM #190: mapOptmization has died

JokerJohn, 2020-06-29:
hello ,thanks for your work!!! when i run lego loam, the mapOptmization  process has died and error shows blow.

 ... logging to /home/xchu/.ros/log/a70e37ea-b9d2-11ea-81c4-309c23ea2da4/roslaunch-xchu-HP-ZHAN-99-Pro-G1-MT-12952.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xchu-HP-ZHAN-99-Pro-G1-MT:43123/

SUMMARY
========

PARAMETERS
 * /mapOptmization/file_dir: /home/xchu/worksp...
 * /mapOptmization/pitch: 0.0
 * /mapOptmization/roll: 1.570795
 * /mapOptmization/x: 0.0
 * /mapOptmization/y: 0.0
 * /mapOptmization/yaw: 1.570795
 * /mapOptmization/z: 0.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: False

NODES
  /
    base_link_to_camera (tf/static_transform_publisher)
    camera_init_to_map (tf/static_transform_publisher)
    featureAssociation (lego_loam/featureAssociation)
    imageProjection (lego_loam/imageProjection)
    mapOptmization (lego_loam/mapOptmization)
    rviz (rviz/rviz)
    transformFusion (lego_loam/transformFusion)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [12978]
process[camera_init_to_map-2]: started with pid [12979]
process[base_link_to_camera-3]: started with pid [12980]
process[imageProjection-4]: started with pid [12987]
process[featureAssociation-5]: started with pid [13001]
process[mapOptmization-6]: started with pid [13005]
process[transformFusion-7]: started with pid [13006]
[ INFO] [1593412700.681740622]: ----> Feature Association Started.
[ INFO] [1593412700.692743490]: ----> Transform Fusion Started.
[ INFO] [1593412700.711314196]: ----> Image Projection Started.
[mapOptmization-6] process has died [pid 13005, exit code -11, cmd /home/xchu/workspace/lego_ws/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/xchu/.ros/log/a70e37ea-b9d2-11ea-81c4-309c23ea2da4/mapOptmization-6.log].
log file: /home/xchu/.ros/log/a70e37ea-b9d2-11ea-81c4-309c23ea2da4/mapOptmization-6*.log

JokerJohn, 2020-08-03:
the reason is the GTSAM version

closed 2020-08-03 by JokerJohn

Ice-SleepingWater, 2020-08-11:
@JokerJohn , Would you mind to share how to fix the gtsam version problem? Thx

JokerJohn, 2020-08-11:
> @JokerJohn , Would you mind to share how to fix the gtsam version problem? Thx

hi, you can change your gtsam version to 4.0.0-alpha2 and have a try

Ice-SleepingWater, 2020-08-11:
> > @JokerJohn , Would you mind to share how to fix the gtsam version problem? Thx
> 
> hi, you can change your gtsam version to 4.0.0-alpha2 and have a try

Thank you for your reply. Actually this is the one I am using right now. I just follow the command of the readme.

Jaskaran-Singh-Sodhi, 2020-08-12:
Possible duplicate of https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/160
Try replacing 4.0.0-alpha2 by the latest version, for example, 4.0.3