RobustFieldAutonomyLab/LeGO-LOAM #194: mapOptmization died

toasttofreedom, 2020-08-04:
hello. when i run roslaunch lego_loam run.launch. there is the same problem as #146 #190 
gtsam version is 4.0.2
```
setting /run_id to b8f792fc-d5fd-11ea-9c9a-f148dc7022ee
process[rosout-1]: started with pid [192986]
started core service [/rosout]
process[rviz-2]: started with pid [192989]
process[camera_init_to_map-3]: started with pid [192993]
process[base_link_to_camera-4]: started with pid [192995]
process[imageProjection-5]: started with pid [192996]
process[featureAssociation-6]: started with pid [192998]
process[mapOptmization-7]: started with pid [193002]
process[transformFusion-8]: started with pid [193004]
[ INFO] [1596509630.636780029]: ----> Transform Fusion Started.
[ INFO] [1596509630.649320095]: ----> Feature Association Started.
[mapOptmization-7] process has died [pid 193002, exit code -11, cmd /home/www/workshop/lego_loam_ws/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/www/.ros/log/b8f792fc-d5fd-11ea-9c9a-f148dc7022ee/mapOptmization-7.log].
log file: /home/www/.ros/log/b8f792fc-d5fd-11ea-9c9a-f148dc7022ee/mapOptmization-7*.log
[ INFO] [1596509630.695966457]: ----> Image Projection Started.
```
then i change my gtsam version to 4.0.0-alpha2. but some new problem arise during catkin_make -j1：
```
In file included from /home/www/workshop/lego_loam_ws/src/LeGO-LOAM/LeGO-LOAM/include/utility.h:21,
from /home/www/workshop/lego_loam_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:34:
/usr/include/pcl-1.10/pcl/filters/voxel_grid.h: In member function ‘std::vector<int> pcl::VoxelGrid<PointT>::getNeighborCentroidIndices(const PointT&, const MatrixXi&) const’:
/usr/include/pcl-1.10/pcl/filters/voxel_grid.h:340:21: error: ‘Index’ is not a member of ‘Eigen’
  340 |         for (Eigen::Index ni = 0; ni < relative_coordinates.cols (); ni++)
      |                     ^~~~~
/usr/include/pcl-1.10/pcl/filters/voxel_grid.h:340:35: error: ‘ni’ was not declared in this scope
  340 |         for (Eigen::Index ni = 0; ni < relative_coordinates.cols (); ni++)
      |                                   ^~

```
Who can tell the reason of this problem.thanks

JokerJohn, 2020-08-11:
hi, you can change your gtsam version to 4.0.0-alpha2 and have a try, i solve this

Jaskaran-Singh-Sodhi, 2020-08-12:
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/190#issuecomment-673033674

toasttofreedom, 2020-08-13:
> hi, you can change your gtsam version to 4.0.0-alpha2 and have a try, i solve this
Thank you for your reply. I have tried. but it didn't work. I have mentioned this above. Fortunately, I solve this problem. The reason why build failed is wrong Eigen's version.The gtsam-4.0.2/3rdparty/eigen version is 3.3.x. It creates a conflict with lego_loam. Then I replace it with gtsam-4.0.0alpha2/3rdparty/eigen which version is 3.2.8. It was finally resolved.

closed 2020-09-19 by stale[bot]