RobustFieldAutonomyLab/LeGO-LOAM #198: mapOptmization process has died

MaxandreOgeret, 2020-09-10:
Hello, I get this error when I start the launchfile :

```
[mapOptmization-7] process has died [pid 6638, exit code 127, cmd /home/maxandre/catkin_pkg/devel/lib/lego_loam/mapOptmization __name:=mapOptmization ].

```

gtsam is at the right version : 4.0.0-alpha2.
How to solve this ? thanks

MaxandreOgeret, 2020-09-10:
Solved when adding /usr/local/lib to ld libraries : 

```bash
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
```

shikeqin123, 2020-10-01:
Hi,I have the same problem with you, Can you explain the solution in more detail？ thank you .

MaxandreOgeret, 2020-10-01:
No problem, the problem was that some libraries were missing. They were installed but not available because not in the LD_LIBRARY_PATH environment variable. So adding `/usr/local/lib` directory to the variable solved my problem.

You just have to run this command before running the launch file : 
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib`

Hope it helps !

yinqijun, 2020-10-24:
> Hi,I have the same problem with you, Can you explain the solution in more detail？ thank you .

Have you solved this problem? And I find a solution,just try `sudo apt-get install libparmetis-dev`.Look forword to your reply.Thank you!

shikeqin123, 2020-10-25:
> > Hi,I have the same problem with you, Can you explain the solution in more detail？ thank you .
> 
> Have you solved this problem? And I find a solution,just try `sudo apt-get install libparmetis-dev`.Look forword for your reply.Thank you!

I had solved it, I just install the right bersion of gtsam---**gtsam-4.0.0-alpha2**,   jsut said   at   readme.md

closed 2020-12-01 by stale[bot]