RobustFieldAutonomyLab/LeGO-LOAM #203: imageProjection-died

lwbdegit, 2020-10-12:
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    base_link_to_camera (tf/static_transform_publisher)
    camera_init_to_map (tf/static_transform_publisher)
    featureAssociation (lego_loam/featureAssociation)
    imageProjection (lego_loam/imageProjection)
    mapOptmization (lego_loam/mapOptmization)
    rviz (rviz/rviz)
    transformFusion (lego_loam/transformFusion)

auto-starting new master
process[master]: started with pid [6364]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ed781bf4-0c63-11eb-b7b7-d8cb8ad34973
WARNING: Metapackage "aerial_mapper" should not have other dependencies besides a buildtool_depend on catkin and exec_depends.
process[rosout-1]: started with pid [6377]
started core service [/rosout]
process[rviz-2]: started with pid [6387]
process[camera_init_to_map-3]: started with pid [6402]
process[base_link_to_camera-4]: started with pid [6403]
process[imageProjection-5]: started with pid [6410]
process[featureAssociation-6]: started with pid [6430]
process[mapOptmization-7]: started with pid [6433]
process[transformFusion-8]: started with pid [6441]
[ INFO] [1602490889.847810367]: ----> Transform Fusion Started.
[ INFO] [1602490889.858379364]: ----> Feature Association Started.
[ INFO] [1602490889.878720058]: ----> Image Projection Started.
[ INFO] [1602490889.917040453]: ----> Map Optimization Started.
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append
[imageProjection-5] process has died [pid 6410, exit code -6, cmd /home/imcngx/legoloam/devel/lib/lego_loam/imageProjection __name:=imageProjection __log:=/home/imcngx/.ros/log/ed781bf4-0c63-11eb-b7b7-d8cb8ad34973/imageProjection-5.log].
log file: /home/imcngx/.ros/log/ed781bf4-0c63-11eb-b7b7-d8cb8ad34973/imageProjection-5*.log

closed 2020-11-24 by stale[bot]

zhangziz, 2021-03-03:
@lwbdegit Sorry to bother, I have the same question, have you deal with?

ResearcherYan, 2022-05-06:
Also met the same question. Solved by changing the version pcl from 1.10 to 1.9.

The environment: Ubuntu 20.04 with ROS Noetic (Desktop-full version).
### Instructions
1. Refer to the [pcl offical installation instructions](https://pcl.readthedocs.io/projects/tutorials/en/latest/compiling_pcl_posix.html) to install pcl-1.9. 
    1. If `tar xvfj` don't work out, then use `tar -zxvf` to untar the compressed file.
    2. You can use `cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/home/user/pcl-1.9 ..` to specify where to install your pcl-1.9 library in order to make a difference with the default pcl-1.10. */home/user/pcl-1.9* is your installed place.
    3. Make sure your machine have at least 8GB memories. When comes to the `make -j2` procedure, just don't use all your cpu kernels here, which may be intriguing memery problems if your memory is not big enough.
2. Modify the *CMakeLists.txt* (the path is somthing like *~/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/CMakeLists.txt*).
    1. Add `set(PCL_DIR "/home/ubuntu/pcl-1.9/share/pcl-1.9")` before `find_package(catkin ......)`.
    2. Change `find_package(PCL REQUIRED QUIET)` to `find_package(PCL 1.9 REQUIRED QUIET)`. 
3. Then use `catkin_make` to rebuild the source code.

ResearcherYan, 2022-05-12:
> Also met the same question. Solved by changing the version pcl from 1.10 to 1.9.
> 
> ### Environment
> * Hardware: Raspberry Pi 4B (arm architecture)
> * OS: Ubuntu 20.04 (arm64 version)
> * ROS: Noetic (Desktop-full version).
> 
> ### Instructions
> 1. Refer to the [pcl offical installation instructions](https://pcl.readthedocs.io/projects/tutorials/en/latest/compiling_pcl_posix.html) to install pcl-1.9.
>    
>    1. If `tar xvfj` don't work out, then use `tar -zxvf` to untar the compressed file.
>    2. You can use `cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/home/user/pcl-1.9 ..` to specify where to install your pcl-1.9 library in order to make a difference with the default pcl-1.10. _/home/user/pcl-1.9_ is your installed place.
>    3. Make sure your machine have at least 8GB memories. When comes to the `make -j2` procedure, just don't use all your cpu kernels here, which may be intriguing memery problems if your memory is not big enough.
> 2. Modify the _CMakeLists.txt_ (the path is somthing like _~/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/CMakeLists.txt_).
>    
>    1. Add `set(PCL_DIR "/home/ubuntu/pcl-1.9/share/pcl-1.9")` before `find_package(catkin ......)`.
>    2. Change `find_package(PCL REQUIRED QUIET)` to `find_package(PCL 1.9 REQUIRED QUIET)`.
> 3. Then use `catkin_make` to rebuild the source code.

### Issues with PCL 1.9
It turns out that although PCL 1.9 can help avoid "imageProjection-died" ERROR, another ERROR just appears instead:
```bash
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
```
At the same time with lots of warnings:
```bash
[pcl::VoxelGrid::applyFilter] Invalid filter field name. Index is -1.
```
With the new error, one can't get the map cloud points in the /tmp directory after use `ctrl + c` to terminate *run.launch* (In order to output the map cloud points file, you need to turn on the Map Cloud in the rviz windows before terminate *run.launch*).

### Debug with PCL 1.10 on Raspberry Pi
So I just go back to PCL 1.10 and try to debug why "imageProjection-died" happened.
When I try to use *cout* and set breakpoints to debug the code, I realize that the error comes from line 175 in *imageProjection.cpp*:
```cpp
pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);
```
Specifically, VSCode DEBUG CONSOLE shows that the error is:
```bash
Thread 1 "imageProjection" received signal SIGBUS, Bus error.
```
After searching the error type SIGBUS for a while, I got the feeling that maybe ARM architecture is the problem.

### Successfully compilation on x86 architecture
However, with exactly the same OS version, ROS version, and PCL version(1.10) on my PC which is x86 architecture, I easily compile LeGO-LOAM successfully (all the problems is list in the [blog](https://blog.csdn.net/weixin_44156680/article/details/118070387?spm=1001.2101.3001.6661.1&utm_medium=distribute.pc_relevant_t0.none-task-blog-2%7Edefault%7ECTRLIST%7Edefault-1.pc_relevant_default&depth_1-utm_source=distribute.pc_relevant_t0.none-task-blog-2%7Edefault%7ECTRLIST%7Edefault-1.pc_relevant_default&utm_relevant_index=1)).

### Conclusions
So it turns out that ARM architecture with Ubuntu 20.04 and ROS Noetic just don't work out with LeGO-LOAM.

Leeable, 2022-07-12:
> 

hey bro ,  you can rebuild the pcl 1.11 version, and it works on my  Raspberry Pi 4B (8GB version) !

ResearcherYan, 2022-07-13:
> > 
> 
> hey bro , you can rebuild the pcl 1.11 version, and it works on my Raspberry Pi 4B (8GB version) !

Thanks for your advice. Will try that when Raspberry Pi 4B is needed in the future. Now I just use my PC as the computing device.

yuantianyuan01, 2022-07-13:
@ResearcherYan Thanks for your detailed instruction but I've encountered the same error on ARM platform. I tried to follow your instruction to install pcl 1.11 and modified the _CMakeLists.txt_. However, it seems that the `catkin_make` is still using the old version (pcl 1.10) and `set(PCL_DIR ...)` is not working. Have you ever met this problem?

yuantianyuan01, 2022-07-13:
In fact the cmakecache shows that all paths about pcl are pointing to pcl 1.11 (no matter whether I add `set(PCL_DIR ...)` `find_package(PCL 1.11 ...)` or not) but somehow the compiler still uses 1.10 version. It gives some warnings like 
`Cannot generate a safe linker search path for target transformFusion
  because files in some directories may conflict with libraries in implicit
  directories:

    link library [libpcl_kdtree.so] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /home/yuanty/pcl-1.11/lib`
and errors like
`/usr/bin/ld: warning: libpcl_octree.so.1.11, needed by /home/yuanty/pcl-1.11/lib/libpcl_filters.so, may conflict with libpcl_octree.so.1.10
`
`/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o:(.data.rel.ro._ZTVN3pcl11KdTreeFLANNINS_9PointXYZIEN5flann9L2_SimpleIfEEEE[_ZTVN3pcl11KdTreeFLANNINS_9PointXYZIEN5flann9L2_SimpleIfEEEE]+0x10): undefined reference to `pcl::KdTreeFLANN<pcl::PointXYZI, flann::L2_Simple<float> >::setInputCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> const> const&, boost::shared_ptr<std::vector<int, std::allocator<int> > const> const&)'`

After I uninstalled pcl-1.11 it builds successfully. Also I tried the same code on my x86 PC and got everything OK.