RobustFieldAutonomyLab/LeGO-LOAM #204: [Error] NaN points using new lidar.

cristianrubioa, 2020-10-15:
Hi, I'm using the Quanergy M8 Lidar.

Regarding the parameter settings, I followed up on: https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/161

But at the time of running the bag I have received the following [Error]: **point cloud is not in dense format please remove NaN points first**

*The problem is that Quanergy M8 Lidar appears to have NaN points.* *Something similar happened here: https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/33 but basically i'm in this same situation: https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/33#issuecomment-706821318

How could i fix it? - Looking forward for your reply.

mpkuse, 2020-10-15:
After I cleaned the data to make it flat, I still see the error: 

```
Failed to find match for field 'ring'.
```

Any comments?

w634307272, 2020-10-15:
> After I cleaned the data to make it flat, I still see the error:
> 
> ```
> Failed to find match for field 'ring'.
> ```
> 
> Any comments?

there is a variable called useCloudRIng in utility.h. Turn it to false.

closed 2020-11-24 by stale[bot]