RobustFieldAutonomyLab/LeGO-LOAM #215: Build error on ubuntu 20.04.

CAKGOD, 2020-12-25:
An error occured when I buid the LeGO-LOAM on ubuntu 20.04. The error info is shown below.
`/usr/include/pcl-1.10/pcl/filters/voxel_grid.h: In member function ‘std::vector<int> pcl::VoxelGrid<PointT>::getNeighborCentroidIndices(const PointT&, const MatrixXi&) const’:
/usr/include/pcl-1.10/pcl/filters/voxel_grid.h:340:21: error: ‘Index’ is not a member of ‘Eigen’
  340 |         for (Eigen::Index ni = 0; ni < relative_coordinates.cols (); ni++)
      |                     ^~~~~
/usr/include/pcl-1.10/pcl/filters/voxel_grid.h:340:21: note: suggested alternatives:
In file included from /usr/include/opencv4/opencv2/flann.hpp:48,
                 from /usr/include/opencv4/opencv2/opencv.hpp:65,
                 from /home/cakgod/lego_loam/src/LeGO-LOAM/LeGO-LOAM/include/utility.h:14,
                 from /home/cakgod/lego_loam/src/LeGO-LOAM/LeGO-LOAM/src/transformFusion.cpp:33:
/usr/include/opencv4/opencv2/flann/flann_base.hpp:104:7: note:   ‘cvflann::Index’
`
Some softwares on this system are shown below.  

- ROS noetic
- PCL 1.10.0
- OpenCV 4.2.0
- Eigen 3.7.0

Can you tell me how to solve this problem? Thanks~

Wing-ka-king, 2021-01-06:
Same here. Facing problems with installation on ubuntu20.04 and ros noetic

Wing-ka-king, 2021-01-14:
Forcing cmake to use c++ 14 did build the package for for me, with some warnings. 
Add this to CMakeLists.txt for LeGO-LOAM
```
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
```

LucasWEIchen, 2021-01-23:
Same here, PCL lib is having problem find  Eigen:Index some how?

LucasWEIchen, 2021-01-23:
> Forcing cmake to use c++ 14 did build the package for me, with some warnings.
> Add this to CMakeLists.txt for LeGO-LOAM
> 
> ```
> set(CMAKE_CXX_STANDARD 14)
> set(CMAKE_CXX_STANDARD_REQUIRED ON)
> ```

This does not work for me

closed 2021-02-05 by CAKGOD

ozgudum, 2021-03-12:
Fix here worked for me: https://github.com/ethz-asl/lidar_align/issues/16

a temporary fix that worked for us:
In /usr/include/flann/util/serialization.h at the top change the corresponding lines to:

```cpp
#include "lz4.h"
#include "lz4hc.h"
```

juliangaal, 2021-03-17:
> Same here, PCL lib is having problem find Eigen:Index some how?

Have you been able to fix the issue?

notplus, 2021-04-22:
May be you can modify file `/usr/include/pcl-1.10/pcl/filters/voxel_grid.h` line 340 and line 669 to solve it.
Just use `int` to substitute `Eigen::Index`. It works for me.
old:
`for (Eigen::Index ni = 0; ni < relative_coordinates.cols (); ni++)`

new:
`for (int ni = 0; ni < relative_coordinates.cols (); ni++)`

davizinho5, 2021-05-12:
The solution provided by @notplus worked for me, although the change mast be done in a couple of places. 

Thanks!

shikeqin123, 2021-06-25:
if you don't want to use `int` to substitute `Eigen::Index`,   Specify a lower version of PCL, for exmaple, I use pcl-1.9 for instead, I add `set(PCL_DIR "/usr/local/share/pcl-1.9/PCLConfig.cmake")` to the CMakeLists.txt

JulianKu, 2021-08-11:
@shikeqin123 For me, installing PCL 1.9 parallel to 1.11 and adding `set(PCL_DIR "/usr/local/share/pcl-1.9/PCLConfig.cmake")` to the CMakeLists.txt still results in `Eigen::Index` errors because it is still using PCL 1.11 in pcl_ros (Maybe additional reconfiguration is required?)

However, the `int` substitution proposed by @notplus works for me

LiShiwei18, 2025-01-16:
If you don't want to modify the pcl header file, you can create a new .h file, then add the following content and include this file
```cpp
namespace Eigen{
    using Index = int;
}
```