RobustFieldAutonomyLab/LeGO-LOAM #216: Has someone succeeded in runing LeGO-LOAM with the RS-Bpearl of RoboSense?

nonlinear1, 2020-12-29:
I use RS-Bpearl of RoboSense to run LeGO-LOAM, But I failed!  The ROS give me the following message:
[image]

[image]
the parameters of RS-Bpearl are as follows:
# of lines :32
Horizontal FoV: 360 degree
Vertical FoV :90 degress
Horizontal Resolution: 0.2 degree
Vertical Resolutiion: 2.81 degree
in utility.h
I give the RS-BPearl parameters are as follows:
//RS-BPearl
extern const int N_SCAN = 32;
extern const int Horizon_SCAN = 1800;
extern const float ang_res_x = 0.2;
extern const float ang_res_y = 2.90;//2.81
extern const float ang_bottom = 90;
extern const int groundScanInd = 0;

nonlinear1, 2020-12-29:
I have solved this problem

closed 2020-12-29 by nonlinear1

arjunskumar, 2022-02-21:
@nonlinear1 How did you solve the problem ?

Kilam3811, 2022-02-22:
Same issue here, may I ask how did you solve it? @nonlinear1  Thanks :)

nonlinear1, 2022-02-23:
you should filter the nan value of robot sense

arjunskumar, 2022-02-23:
@nonlinear1  i was able to solve the issue. Thanks

Kilam3811, 2022-02-24:
@nonlinear1 Thanks for your reply, was able to solve it, thanks 👍

lancelonginus, 2023-07-30:
I use the same type, how did you solve it？