RobustFieldAutonomyLab/LeGO-LOAM #218: Problem when running on custome data.

nickle-fang, 2021-01-08:
I use HESAI[https://www.hesaitech.com/zh](url) lidar to collect a rosbag. The rosbag info is as follows:
```
path:        final_change.bag
version:     2.0
duration:    5:31s (331s)
start:       Jan 08 2021 14:16:18.56 (1610086578.56)
end:         Jan 08 2021 14:21:49.73 (1610086909.73)
size:        6.8 GB
messages:    1289
compression: none [1250/1250 chunks]
types:       rosgraph_msgs/Log       [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /rosout              21 msgs    : rosgraph_msgs/Log      
             /rosout_agg          18 msgs    : rosgraph_msgs/Log      
             /velodyne_points   1250 msgs    : sensor_msgs/PointCloud2
```
And the frame_id of the pointcloud is 'velodyne', which is the same as the official dataset.

But when I run `rosbag play final_change.bag  --clock --topic /velodyne_points` I can't get the odom and rviz get error:
[image]

Oyssster, 2021-02-18:
Hello, I meet the same error, have you solved the problem?

juliangaal, 2021-04-23:
It doesn't seem like you are playing imu data!

SiyuanHuang95, 2021-05-25:
# https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/218#issuecomment-825700874

IMU data is not required indeed. I do not think that should be the problem.

closed 2021-07-08 by stale[bot]