RobustFieldAutonomyLab/LeGO-LOAM #219: opencv.h

robotics2020ashwani, 2021-02-14:
Errors     << lego_loam:make /home/ashwani/catkin_ws/logs/lego_loam/build.make.001.log                                                                                                                            
In file included from /home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:34:
/home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/include/utility.h:13:10: fatal error: opencv/cv.h: No such file or directory
   13 | #include <opencv/cv.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
In file included from /home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/imageProjection.cpp:35:
/home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/include/utility.h:13:10: fatal error: opencv/cv.h: No such file or directory
   13 | #include <opencv/cv.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/imageProjection.dir/build.make:63: CMakeFiles/imageProjection.dir/src/imageProjection.cpp.o] Error 1
make[2]: *** [CMakeFiles/mapOptmization.dir/build.make:63: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:458: CMakeFiles/mapOptmization.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:323: CMakeFiles/imageProjection.dir/all] Error 2
In file included from /home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/featureAssociation.cpp:35:
/home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/include/utility.h:13:10: fatal error: opencv/cv.h: No such file or directory
   13 | #include <opencv/cv.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
In file included from /home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/transformFusion.cpp:33:
/home/ashwani/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/include/utility.h:13:10: fatal error: opencv/cv.h: No such file or directory
   13 | #include <opencv/cv.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/featureAssociation.dir/build.make:63: CMakeFiles/featureAssociation.dir/src/featureAssociation.cpp.o] Error 1
make[2]: *** [CMakeFiles/transformFusion.dir/build.make:63: CMakeFiles/transformFusion.dir/src/transformFusion.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1383: CMakeFiles/featureAssociation.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1160: CMakeFiles/transformFusion.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

I am trying to run this on ros noetic, but it is showing the above error?????

Another thing I want to ask is that will it work on ros noetic or do I have to make some changes????

ShivamPR21, 2021-02-16:
The problem may be the changes in [OpenCV library](https://github.com/opencv/opencv/wiki/ChangeLog#22) the complete library was distributed into submodules. 

> If you detected OpenCV and configured your make scripts using CMake or pkg-config tool, your code will likely build fine without any changes. Otherwise, you will need to modify linker parameters (change the library names) and update the include paths.

> It is still possible to use #include <cv.h> etc. but the recommended notation is:

```.cpp
#include "opencv2/imgproc/imgproc.hpp"
```

closed 2021-06-11 by stale[bot]

Nvethrandil, 2024-03-27:
If anyone still has this issue, there is a fork for OpenCV4 here; https://github.com/Self-SLAM-Lab/LeGO-LOAM_opencv4