RobustFieldAutonomyLab/LeGO-LOAM #221: Hesai pandar 40M--No transform to fixed frame [map]

xxp912, 2021-03-02:
I play the bag file which was collected by Hesai laser pandar 40M, and I get nothing in rviz. There is also a error in rviz For frame [base_link]: No transform to fixed frame [map]. TF error: [Could not find a connection between 'map' and 'base_link' because they are not part of the same tree. Tf has two or more unconnected trees.]
I changed the topic in utility.h and those parameters you had mentioned in https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/3.  
I also uncomment line 159 and catkin_make again. Nothing changes.
How can I fix this problem?

closed 2021-03-05 by xxp912

SiyuanHuang95, 2021-05-25:
have you fixed the problem? I add tf trans, but it still did not work.

xxp912, 2021-05-25:
My wechat account:
xxp18500728702

shahaizi0, 2022-07-05:
那行不要注释，把hesailidar的launch里面timestamp_type的默认值改为realtime