RobustFieldAutonomyLab/LeGO-LOAM #223: test.rviz Fails, Even on Sample Data Sets

jfenton888, 2021-04-06:
I have spent the last couple of days trying to figure out why the out of the box repository, using newly downloaded gtsam and sample data sets, fails consistently when trying to run the provided launch file. I am using ROS Melodic 1.14.10 on Ubuntu 18.04.5 (running using VMWare Fusion on a MacBook Pro). After all appropriate dependencies are added and the repo compiled (and a dataset downloaded), I try to run `roslaunch lego_loam run.launch`, followed by `rosbag play *.bag --clock --topic /velodyne_points /imu/data` in another terminal window.
`[rviz-1] process has died [pid 29927, exit code -6, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/mllax8/eece5554_fp/src/LeGO-LOAM/LeGO-LOAM/launch/test.rviz __name:=rviz __log:=/home/mllax8/.ros/log/c84cb23c-9682-11eb-adb4-000c29be7488/rviz-1.log].
log file: /home/mllax8/.ros/log/c84cb23c-9682-11eb-adb4-000c29be7488/rviz-1*.log`
This error can be achieved pretty consistently, either when immediately when running the launch or the subsequently when the rosbag is played. When this occurs it forces the rviz window to close, making it next to impossible to plot anything. Even when it does work, the only thing that can be seen in the plot is trajectory, the CloudPoint2 type sets are never visible. After tracing back to the `master.log` at the reported directory, I am able to find only this error
`[rosmaster.threadpool][ERROR] 2021-04-04 20:52:28,649: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1095, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 898, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 860, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 837, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused`
This issue seems to be related to [#194](https://github.com/TixiaoShan/LIO-SAM/issues/194) and [#572](https://github.com/ros/ros_comm/issues/572), from which I understand this can be traced to a racing condition where publisher and subscriber node creations are happening out of order. 
I also tried to see if topics were being correctly published to by removing rviz from the launch and `rostopic echo`ing, and ran into one of reasons the racing condition is an issue. There are 9 topics which cannot be simply subscribed to after MapOptimization is run in the launch file. This is because many of them only publish if topic.getNumSubscribers() is non-zero, and despite what the documentation on this method suggests, this value will never be updated during the execution of the program. If there are no nodes already subscribed to the topic when the topic is first advertised, no data can ever be published there. 
My current, very inelegant, debugging phase has all 9 of these problem topics subscribed to using `rostopic echo` before trying to `run.launch` (from which I have removed the rviz). I can then run my own rviz and add topics in and see their data exactly as I would expect. I have a feeling the provided run.launch relies on the fact that rviz will subscribe to the topics before they are advertised on MapOptimization, thus allowing them pass the check for subscribers when those topics are to be published, but this doesn't appear to be reliable in practice right now. 
Another thing I had to address was the issue I encountered when trying to open test.rviz as a configuration from within the rviz GUI. The full sequence consisted of opening all 9 topic subscribers (i.e. `rostopic echo /registered_cloud --noarr`), `roslaunch lego_loam run.launch` with the rviz node removed from the launch file, and `rosbag play *.bag --clock --topic /velodyne_points /imu/data`, and then `rosrun rviz rviz`. If I create my own configuration on the spot or open one I made previously it works just fine, but if I try to open the configuration provided in `test.rviz` the window will fail just like it did originally, but this time it gives some kind of error other than being told to check the logs. 
`mllax8@ubuntu:~/catkin_ws$ rosrun rviz rviz
[ INFO] [1617682054.307659938]: rviz version 1.13.16
[ INFO] [1617682054.307747887]: compiled against Qt version 5.9.5
[ INFO] [1617682054.307775153]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617682054.322769331]: Forcing OpenGl version 0.
[ INFO] [1617682055.653558144, 1496951412.717736001]: Stereo is NOT SUPPORTED
[ INFO] [1617682055.653701969, 1496951412.717736001]: OpenGL device: SVGA3D; build: RELEASE;  LLVM;
[ INFO] [1617682055.654440305, 1496951412.717736001]: OpenGl version: 3.3 (GLSL 3.3) limited to GLSL 1.4 on Mesa system.
Gtk-Message: 00:07:39.384: GtkDialog mapped without a transient parent. This is discouraged.
*** stack smashing detected ***: <unknown> terminated
Aborted (core dumped)`
I am unsure what the critical error was in this, but believe this could also have been what was happening when trying to run the unchanged launch file in the first place. One thought is it could be some package conflict, as i had to download gtsim, libboost-all-dev, and vtk just to get things running as much as they do now, but I do not know how I would be able to track that down.

jfenton888, 2021-04-06:
I am also including the output from making gtsam after uninstalling my original (which was not working for the same reasons, but I thought at the time the issue could be to boost being not found the first time around). I don't believe `pthread` or `MKL` issues would be the cause, but clearly there's something missing or conflicting. 
`~/Downloads/gtsam-4.0.0-alpha2/build$ cmake ..
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_SOURCE_ROOT_DIR: [/home/mllax8/Downloads/gtsam-4.0.0-alpha2]
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   serialization
--   system
--   filesystem
--   thread
--   program_options
--   date_time
--   timer
--   chrono
--   atomic
Ignoring Boost restriction on optional lvalue assignment from rvalues
-- Found Intel TBB
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:207 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenMP_C: -fopenmp  
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:207 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenMP_CXX: -fopenmp  
[... 53 lines cut ...]
--   Default allocator              : TBB
-- Packaging flags                                               
--   CPack Source Generator         : TGZ
--   CPack Generator                : TGZ
-- GTSAM flags                                               
--   Quaternions as default Rot3     : Disabled
--   Runtime consistency checking    : Disabled
--   Rot3 retract is full ExpMap     : Disabled
--   Pose3 retract is full ExpMap    : Disabled
--   Deprecated in GTSAM 4 allowed   : Enabled
--   Point3 is typedef to Vector3    : Disabled
--   Metis-based Nested Dissection   : Enabled
--   Use tangent-space preintegration: Enabled
-- MATLAB toolbox flags                                      
--   Install matlab toolbox         : Disabled
--   Build Wrap                     : Enabled
-- Python module flags                                       
--   Build python module            : Disabled
-- ===============================================================
CMake Warning at CMakeLists.txt:519 (message):
  MKL was not found - this is ok, but note that MKL yields better
  performance.  Set GTSAM_WITH_EIGEN_MKL to 'Off' to disable this warning.

-- Configuring done
-- Generating done`

closed 2021-06-11 by stale[bot]

yupeng126, 2023-08-10:
How to slove this problem?