RobustFieldAutonomyLab/LeGO-LOAM #224: problem when compiling lego-loam in ubuntu 20.04

yyajima21, 2021-04-15:
Hi all, I have meet the following error when I compiled the lego-loam package. I followed this post #215 to resolve this issue, but I am still getting the following errors. I am using ubuntu 20.04, ros noetic, and gtsam 4.0.0-alpha2 described in the readme instruction. Could you tell me how to solve this problem? Thank you.

[ 47%] Linking CXX executable /home/rical/catkin_ws/devel/lib/lego_loam/mapOptmization
`/usr/bin/ld:` CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o: in function `gtsam::GenericValue<gtsam::Pose3>::retract_(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&) const':
mapOptmization.cpp:(.text._ZNK5gtsam12GenericValueINS_5Pose3EE8retract_ERKN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEE[_ZNK5gtsam12GenericValueINS_5Pose3EE8retract_ERKN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEE]+0x7d): undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o: in function `mapOptimization::correctPoses()':
mapOptmization.cpp:(.text._ZN15mapOptimization12correctPosesEv[_ZN15mapOptimization12correctPosesEv]+0x414): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization12correctPosesEv[_ZN15mapOptimization12correctPosesEv]+0x45e): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization12correctPosesEv[_ZN15mapOptimization12correctPosesEv]+0x4ac): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o: in function `gtsam::LieGroup<gtsam::Pose3, 6>::localCoordinates(gtsam::Pose3 const&, gtsam::OptionalJacobian<6, 6>, gtsam::OptionalJacobian<6, 6>) const':
mapOptmization.cpp:(.text._ZNK5gtsam8LieGroupINS_5Pose3ELi6EE16localCoordinatesERKS1_NS_16OptionalJacobianILi6ELi6EEES6_[_ZNK5gtsam8LieGroupINS_5Pose3ELi6EE16localCoordinatesERKS1_NS_16OptionalJacobianILi6ELi6EEES6_]+0x58): undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o: in function `gtsam::BetweenFactor<gtsam::Pose3>::evaluateError(gtsam::Pose3 const&, gtsam::Pose3 const&, boost::optional<Eigen::Matrix<double, -1, -1, 0, -1, -1>&>, boost::optional<Eigen::Matrix<double, -1, -1, 0, -1, -1>&>) const':
mapOptmization.cpp:(.text._ZNK5gtsam13BetweenFactorINS_5Pose3EE13evaluateErrorERKS1_S4_N5boost8optionalIRN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEEESB_[_ZNK5gtsam13BetweenFactorINS_5Pose3EE13evaluateErrorERKS1_S4_N5boost8optionalIRN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEEESB_]+0xbc): undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o: in function `mapOptimization::saveKeyFramesAndFactor()':
mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0x1a3): undefined reference to `gtsam::Rot3::RzRyRx(double, double, double)'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0x42a): undefined reference to `gtsam::Rot3::RzRyRx(double, double, double)'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0xdd9): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0xe0e): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0xe43): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0xf0e): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0xf43): undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o:mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0xf78): more undefined references to `gtsam::Rot3::xyz() const' follow
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o: in function `mapOptimization::saveKeyFramesAndFactor()':
mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0x12a9): undefined reference to `gtsam::Rot3::RzRyRx(double, double, double)'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0x138d): undefined reference to `gtsam::Rot3::RzRyRx(double, double, double)'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0x145e): undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
/usr/bin/ld: mapOptmization.cpp:(.text._ZN15mapOptimization22saveKeyFramesAndFactorEv[_ZN15mapOptimization22saveKeyFramesAndFactorEv]+0x1793): undefined reference to `gtsam::Rot3::RzRyRx(double, double, double)'
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o:(.data.rel.ro._ZTVN5gtsam17NoiseModelFactor2INS_5Pose3ES1_EE[_ZTVN5gtsam17NoiseModelFactor2INS_5Pose3ES1_EE]+0x10): undefined reference to `gtsam::NoiseModelFactor::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&) const'
/usr/bin/ld: CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o:(.data.rel.ro._ZTVN5gtsam17NoiseModelFactor1INS_5Pose3EEE[_ZTVN5gtsam17NoiseModelFactor1INS_5Pose3EEE]+0x10): undefined reference to `gtsam::NoiseModelFactor::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/build.make:500: /home/rical/catkin_ws/devel/lib/lego_loam/mapOptmization] Error 1
make[1]: *** [CMakeFiles/Makefile2:11473: LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j1" `failed`

closed 2021-06-11 by stale[bot]

ravirobot, 2023-04-14:
> you need to add this line inside CMakeLists and it will be working fine. `find_package(Boost REQUIRED COMPONENTS timer thread serialization chrono)`

Wow this worked for me. Thanks!

Cyperpace, 2023-09-26:
> you need to add this line inside CMakeLists and it will be working fine. `find_package(Boost REQUIRED COMPONENTS timer thread serialization chrono)`

> you need to add this line inside CMakeLists and it will be working fine. `find_package(Boost REQUIRED COMPONENTS timer thread serialization chrono)`

thank you！