RobustFieldAutonomyLab/LeGO-LOAM #229: How to save map as pcd!

qaisarehman, 2021-06-17:
Hi

Please can some explain me how I can save the pcd file. I have build the code! its working properly. I tried the data set and I also tried my own recorded bag file. I want to save the map as pcd file but unable to figure out! I tried to search the other issues but still failed.

I will appreciate any suggestions.

Thanks!

qaisarehman, 2021-06-17:
I have managed to save the corner surfaceMap and trajectory file! but still the full cloud pcd is missing!

wienans, 2021-06-18:
https://github.com/koide3/LeGO-LOAM-BOR
Look at this friendly Fork. It dumps the Graph into a folder and later you can import it into interactive_slam (https://github.com/SMRT-AIST/interactive_slam) to export the pcd file

qaisarehman, 2021-06-19:
thanks for the link! I'm going to try these!

qaisarehman, 2021-06-19:
> https://github.com/koide3/LeGO-LOAM-BOR
> Look at this friendly Fork. It dumps the Graph into a folder and later you can import it into interactive_slam (https://github.com/SMRT-AIST/interactive_slam) to export the pcd file

qaisarehman, 2021-06-19:
> https://github.com/koide3/LeGO-LOAM-BOR
> Look at this friendly Fork. It dumps the Graph into a folder and later you can import it into interactive_slam (https://github.com/SMRT-AIST/interactive_slam) to export the pcd file

Is there any way we can generate dense cloud using vlp16 data? lego loam produce very low density cloud!

closed 2021-06-19 by qaisarehman

wienans, 2021-06-19:
How do you define high density? So we use vlp16 too and needed to subsample the pointcloud afterwards to 0.05 cm. So density before was higher.

i would assume covering the same area multiple times with one recording would increase the density, as you Never will exactly hit the same spot and therefore points will „pop up“ in the „free“ spaces to increase density. But this is only a guess

qaisarehman, 2021-06-19:
Actually I want to use the point cloud data for 3d modeling. If you can see the attached image file. It a very large hall! and I used lego loam per produce the cloud! but its very low. I can recognize the features when im far from cloud! when I come closer its just noise! It there any setting in the lego loam which can improve the quality!

[image]

thanks for help!

wienans, 2021-06-19:
I am only a User too. So I don’t know about a Feature like this, I used the Standard config. As I understand Mapping, you have your recorded cloud at each timestep and it will construct a graph in such a way that the pointclouds of each timestep are aligning. So if you feed in more data you will get a longer graph and therefore a more dense cloud map, because of more individual pointclouds.

If you only concerned about the visualization, did you try different tools to vizualite the PCD file?

but in general I think you need more data.

qaisarehman, 2021-06-19:
thanks for the feedback! I also think its because of the low points captured by the sensor. Its recorded in 1 pass. Maybe if I travel back and forth more it will produce more data.  PCD file is very small size as compare to the area!!