RobustFieldAutonomyLab/LeGO-LOAM #234: Loop closure not working

konnoeric, 2021-07-19:
According to the result shown in the photo below, the loop closure is not working. What kind of parameter in the code should I modify?
[image]

BeomBeom2, 2021-07-25:
Hello, I'm sorry for asking a question that has nothing to do with this question. I'm asking because I can't find an answer anywhere. I want to visualize the point cloud like the picture above, but my Rviz only shows the progress path. I want to loam by executing the bag file, but the point cloud does not come out, only the progress trajectory. Do you have any special settings?

[image]
`As an additional question, if I change the fixed frame to aft_mapped, an error appears in Surface features, Edge features as shown below. There was an answer to change the time zone, so my current location is Korea, but I changed the time zone to UTC, but the result was the same.`

[image]

konnoeric, 2021-07-28:
Sorry for being late. I don't quite understand what is the problem. Are you using velodyne VLP-16 ? Because this requires velodyne pointclouds. I will show you my screen on Rviz. 
[image]

After launched the program, I just published the velodyne's point cloud and did nothing. I did not touch any settings nor buttons of Rviz.

BeomBeom2, 2021-07-28:
Thank you so much for your reply.
First of all, the above picture was executed in the order shown on the github page without any setting as a Bag file. In case of map, it works normally, but point cloud is not visualized.

Actually I want to do Lego-Loam with real lidar sensor. The lidar sensor I am using is the VLP-32C model. I changed the parameters of utility.h as shown on the homepage. The model doesn't seem to be causing the error. Both the result of visualization with the actual lidar sensor and the visualization with a Bag file show the same error as in the picture above. I get an error saying For frame [/camera]: no Transform to fixed frame [aft_mapped]. When I searched the error contents, there was a recommendation to update the time on the desktop, so I tried it, but the result was the same. Please let me know if there is a way to fix this error or if there is something I missed.

konnoeric, 2021-07-28:
How about using my bag file? This is VLP-19 of velodyne. If it works with this, then I think that your configuration for your VLP-32C is not working.

[bag file](https://drive.google.com/file/d/1xdVmNby_T7DaEz6hXNJL2ZPHqpQyN8v-/view?usp=sharing)

BeomBeom2, 2021-07-28:
Thanks for providing the file. I will try to it

konnoeric, 2021-07-30:
Sorry I did not notice that. I added you so you may see itt now.

BeomBeom2, 2021-08-01:
Hello. Thanks again for your kind reply. I ran it as a bag, but still the point cloud was not visible in my Rviz. This seems to be my problem. So, I rebuilt it in the same virtual machine afterwards and tried running it again. Then I found out that it became a visualization. However, I tried Lego_loam to get the value from the actual lidar sensor, but an error like velodyne passed to lookuptransform bla bla as below occurred. Do you have any knowledge or advice you can give?

[image]

konnoeric, 2021-08-01:
Sorry I do not have any knowledges to deal with this issue. I don't know.

BeomBeom2, 2021-08-03:
i See. thanks for the reply though

konnoeric, 2021-08-08:
Is this the screenshot of the map created? Im not using robot I just mounted VLP-16 on a handcart and pushed by hand. If I hold the sensor by hand, the result will be horrible since this algorithm is vulnerable to Z-axis drift. I didnt use IMU by the way.

BeomBeom2, 2021-08-15:
Oh that's right, it seems I'm really vulnerable to the z-axis from testing. However, in the bag file you sent, a large amount of the point cloud is visualized at a time, but when I tried, only a very small amount is visualized little by little. also When checking the bag file you sent me, how can the point cloud initially express the right amount of points clearly? 
I will upload the file for your reference.
[My bag file](https://drive.google.com/file/d/1Y8iQgLJEJS85lwh5VRwNEdvPrxtnY8__/view?usp=sharing)

konnoeric, 2021-08-16:
Okay, I will take a look by this weekend.

BeomBeom2, 2021-08-17:
Because you were, I can learn more. Thank you again once :)  :+1:

konnoeric, 2021-08-22:
Hi, these are the results from your bag file with different angles. 

[image]
[image]

[image]
[image]

konnoeric, 2021-08-22:
Actually I modified some parameters for better loop closure in the file below. I think the result is different from yours  because I modified this file. Fell free to try this to have the same settings with me and don't forget to rename the extension as .h, not .txt. I did this because I could not upload this file with the name utilitu.h so I changed it to utility.txt. This file is in the path as you can see in the screenshot.
[image]

[utility.txt](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/files/7026992/utility.txt)

BeomBeom2, 2021-08-23:
Oh thank you for the detailed photo. I have one question, When I made the ros bag file, I recorded the bag file after fixing sensors location.
but visualization is not fixed the location. is it my own error??

```
For reference, I am using the VLP32C version. So I am using the following settings in utility.h.

extern const int N_SCAN = 32;
extern const int Horizon_SCAN = 1800;
extern const float ang_res_x = 0.2;
extern const float ang_res_y = 40.0/float(N_SCAN-1);
extern const float ang_bottom = 25.0;
extern const int groundScanInd = 20;
```

BeomBeom2, 2021-08-25:
oh my jesus, I found why the pointcloud is arbitrary. I think beacuse my room is so small and sealed structure, when I tried to visualization. returned pointcoud is so many that the Lidar sensor wouldn't accept all PCL and then misleaded.

By the way, Did you find a way to make the loopcloser work??

closed 2022-01-09 by stale[bot]

GoldenBoylife, 2023-03-14:
> [konnoeric](/konnoeric)

I have same issue. did you solve the loop closing? 
In the large scale map (staven university bag file), I found loop closing not working. 
it has   z drift.
[image]
<total map: bird eye view>
[image]
<loop closing not working >

ej-sys1, 2024-03-13:
> Actually I modified some parameters for better loop closure in the file below. I think the result is different from yours because I modified this file. Fell free to try this to have the same [...]

Hello, I'm experiencing the same issue. I tried to download the utility.h file, but it says it's expired, and it won't download it. Can I get the file shared again?