RobustFieldAutonomyLab/LeGO-LOAM #245: The imported target "vtk**" references the file    "/usr/bin/***" but this file does not exist。Ubuntu 20

SYLJSTK, 2021-11-19:
When I compile the program,I tried to add the soft connection of VTK, but I just deleted the VTK item. The others can't be found in my system

The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

yongchao987, 2022-04-11:
I've had the same problem recently.

SYLJSTK, 2022-04-11:
> 

I can not solve the prolem.  You can try the Ubuntu 18

L-Reichardt, 2022-07-18:
Issue (possible bug) with how VTK is named on Ubuntu 20. Delete your /devel /build folders and navigate to `/usr/bin/vtk`. Run the following command to copy the vtk7 file with a new name:

```
sudo cp ./vtk7 ./vtk
```
Then compile:

```
catkin_make -j1
```

Should work now. If there are any other issues with Ubuntu 20, refer to [this Issue](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/247).

yinhaosen, 2022-08-18:
@L-Reichardt 's solution works. The issue comes from different naming of vtk on ubuntu. It should be /vtk but /vtk7 (with the version) on the machine. You could also try creating a link like `sudo ln` command.

shrinivas96, 2023-06-07:
I have the same set of warnings as @SYLJSTK, but for a different library (PCL). Creating a soft link to `/usr/bin/vtk7` as `/usr/bin/vtk` removes that one warning. There are other references to files such as the `vtkParseOGLExt-7.1` or `pvtk` which do not even exist in my `/usr/bin/`. I assume it should be as simple as installing some packages which will then have these binaries present in `/usr/bin/`. But I am not sure exactly what should I install? A simple `apt-cache search` does not show any packages named `pvtk` or `vtkParseOGLExt-7.1`. 

Any help regarding this would be very appreciated. Many thanks in advance :)

Vicktoria13, 2023-11-14:
did someone manage to resolve this problem ? I also have this pvtk error and others...

anath93, 2023-12-22:
Did anyone resolve this issue ?

[image]

hitbuyi, 2024-05-03:
I have the same problem
```

-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.
```

anath93, 2024-05-06:
@hitbuyi I resolved this by building the entire kernel again, in my case I was using custom real time kernel. But you can re-image your controller which might be easy.

emmm803, 2025-01-08:
> @hitbuyi I resolved this by building the entire kernel again, in my case I was using custom real time kernel. But you can re-image your controller which might be easy.

May I ask what “building the entire kernel again” means?  Does it refer to “build linux kernel again?” or "build pcl vtk again?"

emmm803, 2025-01-08:
> I have the same set of warnings as @SYLJSTK, but for a different library (PCL). Creating a soft link to `/usr/bin/vtk7` as `/usr/bin/vtk` removes that one warning. There are other references to [...]

Did you ever solve this ? I have the same problem：“vtkParseOGLExt”，“vtkRenderingPythonTkWidgets”，“pvtk” does not exist......

nov05, 2025-04-09:
In `WSL2 Ubuntu 20.04`, building `OpenCV Bridge` from source, ran into the same problem.  
`/usr/bin/vtk7` is there. Other 3 targets are missing.  

```sh
robond@guido:~/vision_opencv/cv_bridge/build$ sudo find / -type f -name "*vtkParseOGLExt*" 2>/dev/null
/usr/lib/cmake/vtk-6.3/Modules/vtkParseOGLExt.cmake
/usr/lib/cmake/vtk-7.1/Modules/vtkParseOGLExt.cmake
^C
robond@guido:~/vision_opencv/cv_bridge/build$ sudo find / -type f -name "*vtkRenderingPythonTkWidgets*" 2>/dev/null
^C
robond@guido:~/vision_opencv/cv_bridge/build$ sudo find / -type f -name "*pvtk*" 2>/dev/null
^C
```

---

```sh
robond@guido:~/vision_opencv/cv_bridge/build$ cmake ..   -DCMAKE_INSTALL_PREFIX=/usr/local   -DPYTHON_EXECUTABLE=$(which python3)   -DPYTHON_INCLUDE_DIR=$(python3 -c "from sysconfig import get_paths; print(get_paths()['include'])")   -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.8.so   -DBoost_NO_BOOST_CMAKE=ON   -DBoost_PYTHON_LIBRARY_RELEASE=/usr/lib/x86_64-linux-gnu/libboost_python38.so   -DOpenCV_DIR=~/opencv-python/_skbuild/linux-x86_64-3.8/cmake-install/lib/cmake/opencv4
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- Using CATKIN_DEVEL_PREFIX: /home/robond/vision_opencv/cv_bridge/build/devel
-- Using CMAKE_PREFIX_PATH: /home/robond/catkin_ws3/devel;/opt/ros/noetic
-- This workspace overlays: /home/robond/catkin_ws3/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/robond/vision_opencv/cv_bridge/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable version "3.8.10", minimum required is "3.8") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robond/vision_opencv/cv_bridge/build
```