RobustFieldAutonomyLab/LeGO-LOAM #247: LeGO-LOAM with gtsam 4.1.1, with ROS-noetic, and Ubuntu 20.04.03

L-Reichardt, 2022-02-04:
Hello all,
I'm posting this as a note for others who might have the same issue in the future. I had issues getting LeGO-LOAM to run with the setup mentioned in the title. I had the following problems which were solved by multiple posts in these "Issues". Summarizing the solutions here.

Symptoms:
- Sometimes issue using cmake to install gtsam 4.0.0-alpha
- Lib Boost not found by cmake
- Nodes crashing during roslaunch

- Cause 1: Gtsam 4.0.0-alpha apparently has compatibility issues with newer versions of the Eigen Library
- Cause 2: CMake is outdated for C++14
- Solution 1: Installing gtsam 4.1.1., adding compatibility into CMake file (CMake flag `-DGTSAM_USE_SYSTEM_EIGEN=ON`)
- Solution 2: Modifications to update CMake file (CMake flag `-lboost_system`, include dirs, link dirs, target_link dirs)

Maybe this helps someone. I used `$ cmake -j1` with success. LeGO-LOAM runs fine now for me.
[CMakeLists.txt](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/files/8004017/CMakeLists.txt)

L-Reichardt, 2022-02-04:
Solved

closed 2022-02-04 by L-Reichardt

MarzanShuvo, 2022-02-14:
hi, I need some helps to use velodyne vlp-16 with Lego-loam. I am using same ros-noetic and ubuntu 20.04. Would you like to give me some suggestions?

L-Reichardt, 2022-02-14:
@MarzanShuvo 
what exactly could I maybe help you with? I am using a different LiDAR, but the principle should remain the same.

MarzanShuvo, 2022-02-14:
@L-Reichardt I am using VLP-16 lidar. I have build the package. While I was running the launch file, rviz was looking like the image bellow. Should I need to change anything in launch file to do the slam? I am not using any imu sensor with it. would you like to give me some guidlines
[image]
?

L-Reichardt, 2022-02-15:
@MarzanShuvo what are the errors in the "Display" section of rviz? Is your topic name the same as in /include/utilities.h? If not, change the topic name in that file.

MarzanShuvo, 2022-02-16:
I think it is same as I am getting the point cloud data like the image bellow. I am also adding frames.
[image]
[frames.pdf](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/files/8077292/frames.pdf)

I have tried to add 
`` in launch file. But the error is consistent.

MarzanShuvo, 2022-02-16:
it is worth to mention that I am getting an error 
**Failed to transform from frame [/camera] to frame [map]**

L-Reichardt, 2022-02-18:
@MarzanShuvo I haven't encountered that error and I was looking a little bit into the code but havent found anything to see where this could come from. Things you could check:
- Did Cmake compile everything correctly? Personally as mentioned in the 1st Post in this thread I had to make changes to get it to run on Ubuntu 20.04.
- Do the sample files provided in this repository work?
- Is Legoloam/ROS actually publishing the messages that show an error in RViz (can be checked with "rostopic hz" or "rostopic echo")
- Fixed frame (in Global Options) and reference frames in RVIZ are correct.

I wish you luck and success in solving your problem.

zxiaomzxm, 2022-02-18:
> it is worth to mention that I am getting an error **Failed to transform from frame [/camera] to frame [map]**

same error here, just replace all /camera to camera and /camera_init to camera_init and all things get right ^ ^

MarzanShuvo, 2022-02-19:
@zxiaomzxm yeah.. It is working perfectly!! Thank you.

aladin2022, 2022-02-24:
hi,I had issues getting LeGO-LOAM to run with the setup mentioned in the title. , i had successfully built the project and installed the libmetis library but still have  this error 
[mapOptmization-7] process has died [pid 28534, exit code 127, cmd /home/aladin/outdoorslam_ws/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/aladin/.ros/log/9e0ac2b2-9559-11ec-a76b-05c4747b78f6/mapOptmization-7.log].
log file: /home/aladin/.ros/log/9e0ac2b2-9559-11ec-a76b-05c4747b78f6/mapOptmization-7*.log

L-Reichardt, 2022-02-25:
@aladin2022 I had the same issue, however for me it was fixed with the changes in the Cmake file (attached above in the issue). For me it was caused by gtsam problems with the Ubuntu 20. eigen library.

samsdolphin, 2022-03-20:
> @aladin2022 I had the same issue, however for me it was fixed with the changes in the Cmake file (attached above in the issue). For me it was caused by gtsam problems with the Ubuntu 20. eigen library.

man, you save my time, thank you!

samsdolphin, 2022-03-20:
> @aladin2022 I had the same issue, however for me it was fixed with the changes in the Cmake file (attached above in the issue). For me it was caused by gtsam problems with the Ubuntu 20. eigen library.

Man, you have saved my time, thank you!

huawei-sai, 2022-12-22:
> > it is worth to mention that I am getting an error **Failed to transform from frame [/camera] to frame [map]**
> 
> same error here, just replace all /camera to camera and /camera_init to camera_init in source code and all things get right ^ ^

This was the solution that worked for me to sole the Failed transform issues with RVIZ in ubuntu 20.04 and ROS Noetic

mgrallos, 2023-05-08:
> hi,I had issues getting LeGO-LOAM to run with the setup mentioned in the title. , i had successfully built the project and installed the libmetis library but still have this error [mapOptmization-7] [...]

I had the same issue I already copied the exact Cmake file and then build it again it did build successfully but no map is form

Rotvie, 2023-05-28:
I've made a tutorial on how to get LeGO-LOAM up and running on Ubuntu 20. I'll leave the link to the [video](https://youtu.be/98maiyRRTlY) in case anyone is interested.

Carl-1999, 2024-03-02:
Thank you very much!  It worked!