RobustFieldAutonomyLab/LeGO-LOAM #260: No Update on Map

SreenathRavi, 2023-02-24:
Tried launching run.launch after that played ros bag no update in rviz. tried launching each node individually. one node crashes when other is launched.

slamvil, 2023-04-20:
I have the same problem... I have tried different bag files and I can see the Lidar points in rviz but there is no map...

mgrallos, 2023-05-05:
i encountered the same problem [mapOptmization-7] process has died [pid 9116, exit code 127the node that crash is map optimization also no map isn't updating

seu-ljdong, 2023-05-22:
> i encountered the same problem [mapOptmization-7] process has died [pid 9116, exit code 127the node that crash is map optimization also no map isn't updating

I encountered the same issue as you did. Have you managed to resolve it?

mgrallos, 2023-05-22:
try this sudo apt-get install libparmetis-dev and make sure you have installed the right version of gtsam for ros noetic

hcws, 2023-11-01:
is there someone solve it ？

MohammadAmmargk8497, 2023-11-24:
Still encountering this problem, anyone has any suggesstions?

suayu, 2024-11-05:
Still encountering this problem, anyone has any suggesstions?