RobustFieldAutonomyLab/LeGO-LOAM #265: It's not working with Livox mid-360, even after I am using PointCloud2 format.

naveenkumar2208, 2023-07-09:


LarryDong, 2023-07-20:
Hi, can you provide more detials? I'm using mid360 and meet some problems. I'm not sure if we have the same issue.
Thanks.

JulesThuillier, 2024-02-02:
@LarryDong I would like to use the Livox MID 360 and Livox Hap, were you happy with the results obtained with Lego-LOAM ? Did you use the internal IMU  of the MID360 ? 
And of course if you have to time to share a few explanation on the modifications that would be awesome ! =)

LarryDong, 2024-02-03:
> Hello @LarryDong ! I would like to use the Livox MID 360 and Livox Hap, were you happy with the results obtained with Lego-LOAM ? Did you use the internal IMU of the MID360 ? And of course if you have to time to share a few explanation on the modifications that would be awesome ! =)

Hi, my suggestion is not to use mid360. Here are some findings:
- Mid360's point cloud is really sparse comparing to other lidar, so you need to adjust the parameters a lot which is different due to the scene and your applications;
- You can re-calculate the "ring" information based on vertical angles and save them into different rings. (Here is my code that convert mid360's `CustomMsg` into ros PointCloud2 with `XYZ+Ring` [https://github.com/hust-ddc-slam/mid360-utils/tree/main/msg_converter](https://github.com/hust-ddc-slam/mid360-utils/tree/main/msg_converter)) . However, mid360's vertical FoV is small so there are only 2-5 scans that can detect the ground, making the ground segmentation really difficult; Maybe you could install the lidar up side down (but I never tried) to make more ground points.
- Uisng internal IMU is okay. The only change may be the acceleration scale (for mid360 is in 'g' not 'm/s2').

So in all, do not try to use mid360 if possible   :<
For Livox Hap, no test.

Good luck.