RobustFieldAutonomyLab/LeGO-LOAM #268: Q. How can I convert output of LeGO-LOAM to 2D grid map for path planning?

yyongjae, 2023-09-11:
My question is on the title.
I want to make a 2D(x,y) Grid map for path planning. 
So I checked pointcloud through RVIZ and I think '/segmented_cloud_pure' topic is suitable for my task. (Is this right? It's from imageProjection.cpp)

Then, I wonder that if segmentedCloudPure(pcl:Pointcloud data) had over 10,000 points, what means the sequence of the points?

bestgaomu, 2024-10-02:
> 我的问题在标题上。 我想制作一个 2D（x，y）网格图用于路径规划。 所以我通过 RVIZ 检查了点云，我认为“/segmented_cloud_pure”主题适合我的任务。（对吗？它来自 imageProjection.cpp）
> 
> 然后，我想知道如果segmentedCloudPure（pcl：Pointcloud data）有超过10,000个点，那么这些点的序列意味着什么？

I also encountered this problem. Did you solve it?

yyongjae, 2024-10-02:
@bestgaomu Do you mean convert PC to 2D?

closed 2025-01-31 by yyongjae