RobustFieldAutonomyLab/LeGO-LOAM #275: terminate called after throwing an instance of 'pcl::IOException' what(): : [pcl::PCDWriter::writeBinary] Input point cloud has no data!

kim-hyunse, 2023-12-10:
this exception keep occurs when make a map (ctrl c). how can i fix this error?
An exception has occurred and the final map is not saved.
[image]

kim-hyunse, 2023-12-14:
i solved my issue. i changed code mapOptmization.cpp. in line 755, 756, 757, 783(map saving part), change like this pcl::io::savePCDFileASCII("/tmp/finalCloud.pcd", *globalMapCloud);.
If you change the path of the pcd validly, the exception will be resolved

DeepDuke, 2024-01-01:
It seems `globalMapCloud` may be empty. I save other pcds successfully after commenting this line like :
```
// pcl::io::savePCDFileASCII("/tmp/finalCloud.pcd", *globalMapCloud);.
```

TheSeanParker, 2024-07-16:
that is my answers 
may be help you
https://zhuanlan.zhihu.com/p/709189056

kim-hyunse, 2024-07-17:
Thank you guys!

closed 2024-07-17 by kim-hyunse