RobustFieldAutonomyLab/LeGO-LOAM #284: Adaption to Robosense RS-Lidar M1

hhackbarth, 2024-04-16:
I am using a Robosense RS-Lidar M1 array LiDAR, which covers 120° HFoV, 25° VFoV, both with 0.2° angular resolution.
The point layout is described here: https://github.com/RoboSense-LiDAR/rs_driver/blob/main/doc/howto/18_about_point_layout.md#1842-mems-lidar

I tried with an adaption of the utility.h file like this:

```
// Using velodyne cloud "ring" channel for image projection (other lidar may have different name for this channel, change "PointXYZIR" below)
extern const bool useCloudRing = false; // if true, ang_res_y and ang_bottom are not used

// RS-Lidar M1
extern const int N_SCAN = 126;
extern const int Horizon_SCAN = 625;
extern const float ang_res_x = 0.2;
extern const float ang_res_y = 0.2;
extern const float ang_bottom = 12.5;
extern const int groundScanInd = 10;
```

The resulting point cloud has similarities to the expectation in the beginnung, so the parameters do not seem to be completely wrong. But the more I move, the wronger it gets (angles of room walls rotated).

Any ideas for a better parametrization?

sunshihao1010, 2024-08-04:
Have you successfully run lego-loam with RS-M1? Could you please tell me your parameter configuration and code modification? Thank you very much

Xiao-Rou, 2024-08-07:
I have the same question , have you successfully run lego-loam with RS-M1? Could you please tell me your parameter configuration and code modification? Thank you very much ^.^